goodrobots/maverick-api #5: Adding schema support for messages is not simple enough

SamuelDudley, 2018-02-03:
Create a mavros to graphql schema generator

SamuelDudley, 2018-02-03:
some mavros messages: https://github.com/mavlink/mavros/tree/master/mavros_msgs/msg
unsure where all the other messages are defined...

fnoop, 2018-02-05:
Notes on schema generation (reflection):
https://github.com/timothyjlaurent/auto-graphene-django
https://github.com/graphql-python/graphene/issues/540
graphene-sqlalchemy and graphene-django are essentially dedicated reflection libraries for those backends.  we may want to create graphene-ros library to do the same.

Also note we should allow introspection by the client:
https://github.com/graphql-python/graphene/issues/3

fnoop, 2018-02-05:
What we want to do is essentially build a ROS ORM for graphene, which sounds like a pretty major job.  However ROS already has excellent python support, so it should be feasible.

SamuelDudley, 2018-02-06:
Re introspection. We could serve the json back to the client from an API endpoint e.g. `/schema`

SamuelDudley, 2018-12-21:
This has become easier with graphql-core-next but there are still lots of opportunities to simplify adding graphql messages