goodrobots/maverick-api #9: CORS support needed

fnoop, 2018-02-15:
```
2018-02-15 21:59:15,218 WARNING 403 GET /subscriptions (127.0.0.1) 0.58ms
2018-02-15 21:59:15,219 DEBUG Cross origin websockets not allowed
```
CORS is needed if our web GCS will support multiple endpoints, by definition any additional endpoint will be cross origin.

fnoop, 2018-02-15:
Added method to GQLSubscriptionHandler to turn off cors checks:
```
    def check_origin(self, origin):
        self.CORS_ORIGINS = ['localhost', 'www.example.com']
        parsed_origin = urlparse(origin)
        # return parsed_origin.hostname in self.CORS_ORIGINS
        return True
```
It currently returns True which disables CORS checks regardless, but should check the hostname against CORS_ORIGINS which should be taken from the config file.
CORS checks can be disabled if we add auth to the websockets

fnoop, 2018-02-15:
Websockets OK but query endpoint rejecting:
```
Failed to load http://www.maverick.one/maverick-api/graphql: Response to preflight request doesn't pass access control check: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://dev.maverick.one' is therefore not allowed access. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.
vue-apollo.esm.js?ebd5:2671 Error sending the query 'stateMessage' TypeError: Failed to fetch
```

SamuelDudley, 2018-02-16:
https://stackoverflow.com/questions/35254742/tornado-server-enable-cors-requests

fnoop, 2018-02-16:
I've basically turned off origin checks for websockets and http requests - check the commits referenced, but this should probably stay open as a marker to investigate CORS and ramifications more closely.
I've added support for auth in the client which we should look into supporting at the api end, will raise a separate issue.

fnoop, 2020-03-26:
Getting CORS errors in firefox - works fine in chrome:
[image]

SamuelDudley, 2020-03-31:
Looks like an issue with chrome. As far as I can understand the preflight options call is made without the cert in Firefox but not chrome. Reading the above thread there may be a nginx setting that needs changing.

https://bugzilla.mozilla.org/show_bug.cgi?id=1019603

SamuelDudley, 2020-03-31:
ssl_verify_client on -> ssl_verify_client optional

fnoop, 2020-03-31:
These requests aren't going through nginx - they're direct to -api

SamuelDudley, 2020-04-01:
problem appears to be a certificate issue. I too had this problem with firefox however once the certificate was installed correctly all requests via https work as expected (linux + firefox). Note webrtc video does not work but that appears to be a browser compatibility issue rather than anything else.

SamuelDudley, 2020-04-01:
[image]