goodrobots/maverick-api #10: Parameter support

SamuelDudley, 2018-02-16:
Add graphql support for getting and setting parameters via mavros

SamuelDudley, 2018-02-17:
Added getter support 
https://github.com/goodrobots/maverick-api/commit/b89dd950b526273805a603d8d9181dfdb3d26908 and https://github.com/goodrobots/maverick-api/commit/de852dc51f3f07f8b415df3254265851bf86b4ab

fnoop, 2018-02-19:
Is it possible to sort the param query response alphabetically?

fnoop, 2018-02-19:
It's possible to sort the params at the client end by creating a computed copy of the immutable query array and iterating through the sorted copy:
```
computed: {
    sortedParams () {
      return this.params.slice().sort(function (a, b) {
        if (a.id < b.id) {
          return -1
        } else if (a.id > b.id) {
          return 1
        } else {
          return 0
        }
      })
    }
  },
```
It works but it's clunky and I think this is a good example of logic we should try and keep in the api, so the client has to do as little as possible.  Perhaps we could add a sort query param?  Or in this case just sort by default - I can't think of a situation where you would want unsorted params.
Another downside of doing it in the client as above, is whenever the subscription updates the original query in the graphql client, I suspect it probably has to slice and sort the array all over again.

SamuelDudley, 2018-02-19:
@fnoop params are now returned sorted https://github.com/goodrobots/maverick-api/commit/bcdb62267111e198a6f3dc70b3dad4aaf96d69c8

fnoop, 2018-02-19:
Thanks!  Is fixing the subscriptions an easy thing?

SamuelDudley, 2018-02-19:
fixed subscriptions with https://github.com/goodrobots/maverick-api/commit/baff25e7fd8b0abb86d5b56f093d7d6986c84a5c

fnoop, 2018-02-19:
Is it possible to support meta.fields with a proper schema?  It looks like the fieldnames are fixed.  These fields are very useful to use in the interface, and because the generated js objects from graphql are immutable (they act as state similar to vuex) then I can't translate them to js object in place, rather an deep clone has to be made:
```
for (const param of this.params) {
          let paramcopy = JSON.parse(JSON.stringify(param))
          if (paramcopy.id.match(paramRegexp)) {
            if (paramcopy.meta) {
              if (paramcopy.meta.fields) paramcopy.meta.fields = JSON.parse(paramcopy.meta.fields)
              if (paramcopy.meta.values) paramcopy.meta.values = JSON.parse(paramcopy.meta.values)
            }
            groups[group].push(paramcopy)
          }
```
This is pretty clunky.

fnoop, 2018-02-19:
It seems if you change a parameter in the firmware, you have to restart maverick-api to pick it up.

SamuelDudley, 2018-02-20:
> Is it possible to support meta.fields with a proper schema?

Looking more closely at the variety of options stored in `meta.fields` we see the following options:
```
'fields': {}
'fields': {'Units': 'Hz', 'Range': '1.000 100.000', 'UnitText': 'hertz'}
'fields': {'Range': '1 8', 'RebootRequired': 'True', 'Increment': '1'}
'fields': {'Bitmask': '0:FirstIMU,1:SecondIMU,2:ThirdIMU,3:FourthIMU,4:FifthIMU,5:SixthIMU', 'RebootRequired': 'True'}
```
so we have the following:
 - Units (string)
 - UnitText (string)
 - Range (2 numbers)
 - Increment (number)
 - RebootRequired (bool)
 - Bitmask (ints mapped to strings)

The ones that stand out are `Range` and `Bitmask` which are not core types. I'm assuming based on your above message that further processing of those fields is required so that you can avoid the deep copy. The `Bitmask` could be possibly mapped to an enum type in graphene, while the range might be best as an enum type containing `min` and `max` fields.

If that all works then I can also represent `meta.values` as an enum type in graphene rather than the JSON.
It's going to take me a while to work out how to do the above, but ultimately it seems required.
Will that work for you?

SamuelDudley, 2018-02-20:
> It seems if you change a parameter in the firmware, you have to restart maverick-api to pick it up.

This would be true. There is nothing in the API listening for parameter changes originating from anywhere other than the graphql client via the mutation process. I need a ros node to subscribe to parameter changes.

SamuelDudley, 2018-02-20:
Another thing to think about is loading and saving params to / from file... Will the browser just pass the file to the api or will the browser read the file and make mutations to the api?

SamuelDudley, 2018-02-20:
To support notification of parameter changes in the mavros state we will need to modify:
https://github.com/mavlink/mavros/blob/master/mavros/src/plugins/param.cpp
https://github.com/mavlink/mavros/blob/master/mavros/src/mavros/param.py

We could add a listener in maverick-api and listen to a new handler after the `set_value()` method call. The handler would need to return the parameter (value and name).

SamuelDudley, 2018-02-27:
Notification support added from mavros. So external changes (via QGC for example) will now update maverick-web

# param meta
We need to support a local and remote option. If we set the param meta mode to 'remote' we:
1. Try to download and save the autotest xml file from ardupilot (for all vehicles)
2. If we cant download the autotest xml we use the last saved copy on the api disk.
3. Failing the above steps or if local mode is selected we grab the meta from the ardupilot codebase on the api disk (have yet to test this or get it working)

SamuelDudley, 2018-03-07:
Vehicle / AP auto detect and param parse added with https://github.com/goodrobots/maverick-api/commit/90259f9abf28e4b3c693a9b579db20cf4b54e2ba

fnoop, 2019-02-08:
Params need updating for new -api design.
If the subscription could just return changed parameters, instead of all parameters, that would be really good.  I think previously subscription updates returned all parameters and it was killing the front-end having to parse these - particularly as some parameters update quite frequently.

SamuelDudley, 2019-09-14:
Now we have the basics working again, begin to support param meta