goodrobots/maverick-api #12: Add support for users and optional auth

SamuelDudley, 2018-02-16:
User login
Https and secure web sockets
Retrieve user preferences from db
Support oauth?

fnoop, 2018-02-16:
Related: https://github.com/goodrobots/maverick/issues/689
I had thought creating a central LDAP server for each maverick computer that everything else could connect to.  oauth etc has the problem that they are inherently internet-y services, although they might be a really good idea as an option?

SamuelDudley, 2018-02-17:
Agreed that oauth isn't particularly useful for non connected applications. I don't have a use case in mind atm. LDAP looks good! 

https://code.tutsplus.com/tutorials/flask-authentication-with-ldap--cms-23101

SamuelDudley, 2018-02-17:
https://github.com/gmr/Tinman/blob/master/tinman/auth/ldapauth.py

https://github.com/gmr/Tinman/blob/master/examples/helloworld_basic_ldap.py

SamuelDudley, 2018-02-17:
@fnoop I'm assuming we will have a form on the web UI that when submitted (https or wss) is handled by the API to check credentials against the LDAP server. 
The other option is to have the web UI / server connect to the LDAP server directly.

SamuelDudley, 2018-10-01:
One time login lib that may be of use: https://github.com/pyotp/pyotp

SamuelDudley, 2020-02-25:
https://github.com/casbin/pycasbin

fnoop, 2020-02-25:
I've been looking at ldap in the last couple of days.  It's really horrible to work with directly, I'm not sure we want to go through the pain.  But it has the advantage of being very well supported by lots of apps, platforms etc, from ssh to web servers to web apps to (python) middleware.  'IAM' (Identity and Access Management) has taken over as the new buzzword, which encompasses ldap and adds kerberos and all sorts of other protocols and subsystems, usually with a nice web interface, to make identity management an easier service to install and manage.  The trouble is most of these are much heavier, often written in Java, which aren't suitable for the smaller platforms we generally target.  However a lot of them are opensource, eg:
https://www.gluu.org/

I suggest that we don't think too hard about this for now - just get a very basic auth working even if from a sqlite or whatever database.  Or even no auth inherent to the app for now, we can just protect everything using webserver auth.  Or like Maverick itself, just have a single default user (mav) with a default password that can easily be reset.

SamuelDudley, 2020-02-25:
Yep, agreed. I started adding auth hooks to -api some time ago here https://github.com/goodrobots/maverick-api/blob/master/maverick_api/modules/base/tornadoql/session_control.py but was having a bit more of a think about how I want to _actually_ do the authentication. 
For the moment im not too worried.  The auth hooks will be similar(ish) regardless of what we use for the authentication.