goodrobots/maverick-api #14: Parameter value whitespace

fnoop, 2018-02-22:
eg. FRAME_CLASS:
{ "0": "Undefined", " 11": "Heli_Dual", " 12": "DodecaHexa", " 3": "Octa", " 2": "Hexa", " 1": "Quad", " 6": "Heli", " 7": "Tri", " 4": "OctaQuad", " 5": "Y6", " 8": "SingleCopter", " 9": "CoaxCopter" }
The " 1" value has a whitespace at the beginning of the key, so this doesn't then match when trying to display the parameter value.  Can we strip whitespace out of keys?   Not sure if this will break anything else - perhaps strip prefixing or suffixing whitespace?

SamuelDudley, 2018-02-22:
Simples! will do.

SamuelDudley, 2018-02-24:
closed with https://github.com/goodrobots/maverick-api/commit/57de0d874555c1a32dde0a8e55b1185e388bdb5a

closed 2018-02-24 by SamuelDudley