goodrobots/maverick-api #15: Parameter group

fnoop, 2018-02-22:
In the parameter group screen (http://www.maverick.one/maverick-web/#/config/parameter-groups), the params are grouped by the first _ split component of the parameter ID (eg. ACCEL_Z_D becomes ACCEL).  It's very difficult to do this in the frontend without creating a complete new set of nested arrays, which takes a lot of code and cycles, and then breaks some reactive data links.

If each parameter could have an extra 'group' or 'prefix' field inserted by the API that has this component, it would greatly simplify the frontend.

SamuelDudley, 2018-02-22:
Can do.

SamuelDudley, 2018-02-24:
closed with https://github.com/goodrobots/maverick-api/commit/57de0d874555c1a32dde0a8e55b1185e388bdb5a

closed 2018-02-24 by SamuelDudley