goodrobots/maverick-api #17: Mission support

SamuelDudley, 2018-02-24:
Store more than one mission to allow selection

Load missions from file (XML and non XML format)
Rec mission msg from mavros
Send mission msgs via mavros

Report missions to client via graphql
Update missions via graphql mutations

fnoop, 2018-02-24:
Perhaps a different message for the FC active waypoints, and stored missions?
Or if done in a single message then a meta field (boolean) and/or fixed name that denotes the active FC waypoints.  A query parameter could easily be used then to retrieve the active waypoints.
It would be great to easily plan, store and retrieve/activate lots of different missions, something I've always found very clunky on traditional GCSs.
We'll need to start the sqlite database storage for this, I guess.

SamuelDudley, 2018-02-24:
I agree with all of your comments and suggestions. Supporting multiple missions and the ability to stack them or swap between them is a requirement IMO. I need to think a little more about how the mission system could work, but I would like to abstract / simplify mission planning and generate complex missions from minimal client inputs at the api level. 

e.g. planing a mapping mission should be simple as picking your sensor+lens+mount from a list, specifying ground sampling density and providing a polygon(s). The api should generate and manage the mission(s) required to map the area.

fnoop, 2019-01-21:
Now handled in #86

closed 2019-01-21 by fnoop