goodrobots/maverick-api #20: Request data streams

fnoop, 2018-02-27:
Looks like maverick-api needs to request the various data streams in order to receive most of the data.  At the moment after you start sitl and maverick-api you get the basic data only - no vfr/gps/pose/imu data.  But if you just connect a GCS like QGC then immediately the rest of the data starts streaming.  This is presumably because the GCS requests all the data streams to start.

SamuelDudley, 2018-02-27:
I guess that we want *all of the data*... I'll see whats involved and get mavros requesting everything

fnoop, 2018-02-27:
I guess so, and the stream rates will be important too.  Will be a balance of getting fast enough updates but not overloading the link/processing.

fnoop, 2018-03-01:
Added Connection.streams() method which does the initial stream requests.  Actual stream values (stream_rates dict) should be set in the json config, somehow.

closed 2019-01-21 by fnoop