goodrobots/maverick-api #33: Altitude data?

fnoop, 2018-04-01:
vfrHudMessage.altitude gives an odd figure - when flying at reported 100m from a GCS, this message field gives about 684 value slightly lower that the navSatFix.altitude:
`vfrHudMessage: 684.22998046875  :: navSatFix: 703.6901112815922`

fnoop, 2018-04-01:
I would have thought vfrHudMessage should be altitude above home?

SamuelDudley, 2018-12-21:
I'm assuming we want alt above home? If so I can send it in a separate message

fnoop, 2018-12-21:
Yes please

SamuelDudley, 2019-01-13:
Forgot to close this when I added rel alt into the vrf hud message
https://github.com/goodrobots/maverick-api/blob/master/maverick_api/modules/api/mavros/__init__.py#L600
https://github.com/goodrobots/maverick-api/blob/master/maverick_api/modules/api/mavros/__init__.py#L151

fnoop, 2019-01-13:
Just a 'meta' query here - by adding relative alt we're drifting away from the mavlink definitions:
https://mavlink.io/en/messages/common.html#VFR_HUD

Does that matter?

SamuelDudley, 2019-01-13:
probably best to send it in the correct mavlink message.