goodrobots/maverick-api #36: Modularise schema

fnoop, 2018-04-17:
Api and Schema is all currently in a couple of big files which makes it difficult to add new modules/features.  Would be much easier if the logic for each area was broken out into separate python modules.

SamuelDudley, 2018-09-18:
This is really important and is high on my list of things to do. Currently I think the schema and the mavros callback functionality should live together in a discreet module.

SamuelDudley, 2018-09-25:
Generate a structure so that tornado IO loop callbacks, mavros callbacks and schema can live in one "module". For now each module will be loaded at runtime, with the plan to add module loading controls at a later date.

SamuelDudley, 2019-01-06:
closing as the new api code has allowed this to move forward. See #55 for further work.

closed 2019-01-06 by SamuelDudley