goodrobots/maverick-api #39: get_vehicle_info error

fnoop, 2018-09-27:
At the beginning of startup, log entry:
`2018-09-27 08:21:04,788 ERROR An error occurred while retrieving vehicle info via ROS: service [/mavros/get_vehicle_info] unavailable`

Doesn't seem to exist:
```
[dev] [mav@goodrobots ~/code/maverick-api]$ rostopic echo /mavros/get_vehicle_info
WARNING: topic [/mavros/get_vehicle_info] does not appear to be published yet
```

fnoop, 2018-09-27:
This comes from maverick-api:
```
def vehicle_info(self):
        # Create ROS service definition for VehicleInfo
        get_vehicle_info = rospy.ServiceProxy(mavros.get_topic('get_vehicle_info'), VehicleInfo)
        try:
            self.info = get_vehicle_info()
            self.meta_string = get_meta_string(self.info)
            app_log.debug(self.info)
            app_log.info(self.meta_string)
        except rospy.ServiceException as ex:
            app_log.error('An error occurred while retrieving vehicle info via ROS: {0}'.format(ex))
```

I can't see any reference to get_vehicle_info anywhere in the mavros code, or find any reference to it anywhere else.

SamuelDudley, 2018-09-27:
output of `rosservice list` on www.maverick.one
```
rosservice list 
/listener_24914_1537973168350/get_loggers
/listener_24914_1537973168350/set_logger_level
/mavros/cmd/arming
/mavros/cmd/command
/mavros/cmd/command_int
/mavros/cmd/land
/mavros/cmd/set_home
/mavros/cmd/takeoff
/mavros/cmd/trigger_control
/mavros/get_loggers
/mavros/get_vehicle_info
/mavros/home_position/req_update
/mavros/log_transfer/raw/log_request_data
/mavros/log_transfer/raw/log_request_end
/mavros/log_transfer/raw/log_request_list
/mavros/mission/clear
/mavros/mission/pull
/mavros/mission/push
/mavros/mission/set_current
/mavros/param/get
/mavros/param/pull
/mavros/param/push
/mavros/param/set
/mavros/set_logger_level
/mavros/set_mode
/mavros/set_stream_rate
/mavros/setpoint_position/mav_frame
/mavros/setpoint_velocity/mav_frame
/rosout/get_loggers
/rosout/set_logger_level
```
You can see that the mavros service `/mavros/get_vehicle_info` exists, so it seems that there are parts of mavros not loading correctly

SamuelDudley, 2018-09-27:
on newdev.maverick.one we get the following

```
rosservice list
/listener_18609_1537347853217/get_loggers
/listener_18609_1537347853217/set_logger_level
/listener_19119_1538086269626/get_loggers
/listener_19119_1538086269626/set_logger_level
/mavros/cmd/arming
/mavros/cmd/command
/mavros/cmd/command_int
/mavros/cmd/land
/mavros/cmd/set_home
/mavros/cmd/takeoff
/mavros/cmd/trigger_control
/mavros/get_loggers
/mavros/home_position/req_update
/mavros/log_transfer/raw/log_request_data
/mavros/log_transfer/raw/log_request_end
/mavros/log_transfer/raw/log_request_list
/mavros/mission/clear
/mavros/mission/pull
/mavros/mission/push
/mavros/mission/set_current
/mavros/param/get
/mavros/param/pull
/mavros/param/push
/mavros/param/set
/mavros/set_logger_level
/mavros/set_mode
/mavros/set_stream_rate
/mavros/setpoint_position/mav_frame
/mavros/setpoint_velocity/mav_frame
/rosout/get_loggers
/rosout/set_logger_level
```

SamuelDudley, 2018-09-27:
looking at the imports `VehicleInfo` comes from the special maverick_mavros module: `from mavros_maverick.srv import VehicleInfo`. So perhaps that has failed to install or start correctly?

fnoop, 2018-09-28:
Oh balls, completely forgot about that!  Should add to the readme as a reminder.  @kiwa21, Do you think it's something worth submitting to mavros for inclusion upstream?

fnoop, 2018-09-28:
The library has built and resolves OK:
```
[dev] [mav@goodrobots ~/var/build/catkin_mavros_maverick]$ ldd ~/software/ros/current/lib/libmavros_maverick.so
	linux-vdso.so.1 (0x00007ffeab581000)
	libmavros.so => /opt/ros/melodic/lib/libmavros.so (0x00007f3d0e9ba000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007f3d0e793000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007f3d0e3f6000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007f3d0e1c0000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007f3d0dfbd000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f3d0ddb8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3d0db99000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007f3d0d994000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3d0d606000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3d0d3ee000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3d0cffd000)
	libroslib.so => /opt/ros/melodic/lib/libroslib.so (0x00007f3d0cdea000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007f3d0cbd6000)
	libtf2_ros.so => /opt/ros/melodic/lib/libtf2_ros.so (0x00007f3d0c92b000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f3d0c711000)
	libtf2.so => /opt/ros/melodic/lib/libtf2.so (0x00007f3d0c4dd000)
	libmavconn.so => /opt/ros/melodic/lib/libmavconn.so (0x00007f3d0c284000)
	librosconsole_bridge.so => /opt/ros/melodic/lib/librosconsole_bridge.so (0x00007f3d0c080000)
	libeigen_conversions.so => /opt/ros/melodic/lib/libeigen_conversions.so (0x00007f3d0be7b000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007f3d0bc5b000)
	libGeographic.so.17 => /usr/lib/x86_64-linux-gnu/libGeographic.so.17 (0x00007f3d0b9a4000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3d0b606000)
	libPocoFoundation.so.50 => /usr/lib/libPocoFoundation.so.50 (0x00007f3d0b25d000)
	libxmlrpcpp.so => /opt/ros/melodic/lib/libxmlrpcpp.so (0x00007f3d0b03f000)
	libcpp_common.so => /opt/ros/melodic/lib/libcpp_common.so (0x00007f3d0ae37000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007f3d0ac12000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007f3d0aa0d000)
	librosconsole_log4cxx.so => /opt/ros/melodic/lib/librosconsole_log4cxx.so (0x00007f3d0a7f2000)
	librosconsole_backend_interface.so => /opt/ros/melodic/lib/librosconsole_backend_interface.so (0x00007f3d0a5f0000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007f3d0a227000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007f3d09f1f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f3d0ee5a000)
	librospack.so => /opt/ros/melodic/lib/librospack.so (0x00007f3d09cdc000)
	libactionlib.so => /opt/ros/melodic/lib/libactionlib.so (0x00007f3d09abb000)
	liborocos-kdl.so.1.4 => /opt/ros/melodic/lib/liborocos-kdl.so.1.4 (0x00007f3d09826000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f3d0961e000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f3d0941a000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f3d091a8000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f3d08f8b000)
	libapr-1.so.0 => /usr/lib/x86_64-linux-gnu/libapr-1.so.0 (0x00007f3d08d56000)
	libaprutil-1.so.0 => /usr/lib/x86_64-linux-gnu/libaprutil-1.so.0 (0x00007f3d08b2b000)
	libicui18n.so.60 => /usr/lib/x86_64-linux-gnu/libicui18n.so.60 (0x00007f3d0868a000)
	libicuuc.so.60 => /usr/lib/x86_64-linux-gnu/libicuuc.so.60 (0x00007f3d082d3000)
	libboost_program_options.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.65.1 (0x00007f3d08052000)
	libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f3d07ad5000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f3d078ce000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f3d07696000)
	libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f3d07464000)
	libicudata.so.60 => /usr/lib/x86_64-linux-gnu/libicudata.so.60 (0x00007f3d058bb000)
	libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f3d056b8000)
```

SamuelDudley, 2018-09-28:
Considering that it compiles I don't see the hpp change helping us out here... Any other ideas?

fnoop, 2018-12-05:
Fixed by migrating to new ros headers with scripts/pluginlib_headers_migration.py

closed 2018-12-05 by fnoop