goodrobots/maverick-api #40: Consider alternate options to graphql

SamuelDudley, 2018-09-30:
Graphql has limitations that may not allow us to achieve the desired project architecture. This issue is to keep track of thoughts and requirements to help decide if we need to change out our API<->Client communication mechanism. 

Required: 

- Support a single client connecting to many API instances simultaneously 
- API instances should not have to communicate between one another for this feature to function
- Non-polling (e.g. support websocket streaming)
- Ability to be modularised 

Desired:

- An interface that can be polled to request capability (introspection)
- Existing libraries that can be leveraged to make out jobs simpler (Client and API) 

Related:
https://github.com/goodrobots/maverick-web/issues/35
https://github.com/Akryum/vue-apollo/issues/216
https://medium.com/open-graphql/apollo-multiple-clients-with-react-b34b571210a5

SamuelDudley, 2018-09-30:
One option is to go back to a websocket using vanilla JavaScript. We can send json from the API and action the data in the client accordingly.

SamuelDudley, 2018-10-02:
@fnoop is doing a deep dive into the graphql client code to see if we are able to achieve the desired project architecture.

@cglusky Suggested that the API could live at the nginx level (e.g. kong and nchan)

fnoop, 2018-12-19:
I think we can close this.  We've proved graphql client can talk to multiple apis simultaneously, and we haven't identified a realistic alternative.

closed 2018-12-19 by fnoop