goodrobots/maverick-api #44: Restructure filetree

fnoop, 2018-12-20:
Currently the code all sits in maverick_api directory, with some meta files up at the top level:
```
-rw-rw-r-- 1 mav mav  891 Dec 19 23:43 Jenkinsfile
-rw-rw-r-- 1 mav mav 1063 Dec 19 23:43 LICENSE
-rw-rw-r-- 1 mav mav  754 Dec 19 23:43 README.md
drwxrwxr-x 9 mav mav 4096 Dec 19 23:44 maverick_api
-rw-rw-r-- 1 mav mav   68 Dec 19 23:43 requirements.txt
```

Inside maverick_api, there are several directories:
```
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 config
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 database
-rwxrwxr-x 1 mav mav 7054 Dec 19 23:43 maverick_api.py
drwxrwxr-x 5 mav mav 4096 Dec 19 23:43 modules
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 param
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 static
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 tornadoql
drwxrwxr-x 2 mav mav 4096 Dec 19 23:43 util
```

It would be nice to flatten this so maverick_api.py sits at the top level, and to simplify/restructure the various directories to be cleaner for end users.  For example, config is where a user would expect to find configuration files, not logic that parses the configuration.

For example, something like:
```
LICENSE
README.md
maverick_api.py
config/
data/
modules/
```
where modules/ contains most of the logic and is further structured to hold -api modules, and the graphql iterated modules.

This simplified structure should be more intuitive for users, and simpler to deploy and develop?

SamuelDudley, 2018-12-20:
Moving my comments from gitter:
While I agree with what you are saying, I recently moved to this structure (from a flatter structure) based on published best practices for python modules. As an example: https://github.com/ArduPilot/MAVProxy and https://github.com/graphql-python/graphql-core-next
note the nested layer these packages have. The top layer is all the meta files needed to build, test and provide info for the project. The nested `maverick_api` directory contains the project source (similar to a `src` folder in a C++ application). It also helps to separate tests from the source code directory: https://docs.pytest.org/en/latest/goodpractices.html (tests are the next thing to come)
My ultimate aim is to have tagged releases pushed to the python package repository so that users can `pip install maverick-api` the current directory structure is to facilitate this. I feel its moving in the correct direction but is not yet 100%

SamuelDudley, 2018-12-20:
To further the discussion above. I think we have to live with the meta files + docs + tests in the first layer.
I'm onboard with changing the file structure within the `maverick_api` directory from what it is (a bit of a mess really) into something more logical. 

- Keeping the `modules` directory at this layer makes good sense for many reasons which you mentioned above.  
 - Splitting out the `config` directory and removing unneeded logic (into a module?) from here also makes sense
 - A `data` folder is a good idea and would currently contain the parameter files and database tables (parameter getting and parsing logic should be split out and form a module)
- Adding a `lib` folder makes sense to me which would contain libraries unrelated to modules, currently this would contain the `tornadoql` folder

 ```
LICENSE
README.md
requirements.txt
Jenkinsfile
tests/
docs/
maverick_api/
   maverick_api.py
   __init__.py
   config/
      config.json
   data/
      param/
      database/
   modules/
   libs/
       tornadoql/
```

fnoop, 2018-12-20:
Sounds great.  Can the modules directory be used for modules other than auto-parsed graphql modules?

SamuelDudley, 2018-12-20:
Yes they can (or will) I'm looking at ways to load the contents of the modules folder at runtime. There are a few options here that make it a little tricky. Non asynchronous code has to be run in a thread or process seperate to the main tornado ioloop (see mavros). This thread is started when maverick api starts but really the modules directory needs to be scanned and all the code needs to auto run (otherwise developers will need to modify the main tornado server code).

SamuelDudley, 2018-12-20:
I will probably add a section to the config that defines what modules are loaded at startup. I've been thinking of ways to support loading and unloading modules at runtime (like mavproxy) but the graphql schema makes that difficult...

fnoop, 2018-12-20:
OK so makes sense then to have a separate folder or subfolder for autoload modules, either:
 - modules/
 - autoload_modules/

modules/autoload
modules/...

I'd go for the latter for tidyness, if it doesn't have a technical impact

fnoop, 2018-12-20:
my 2c - I don't think loading/unloading modules during runtime is important - it's entirely reasonable to restart the api if you're adding/removing modules to the filesystem.

SamuelDudley, 2018-12-20:
I can agree with the loading/unloading at runtime statement, it's mainly useful during development of the module.
Re making a separate folder for the autorun modules, I would prefer to define them in the config file and not have them in a different location. That way the file system is consistent across many installs but the configuration file can change. Thoughts?

SamuelDudley, 2018-12-20:
Say I make a module that dynamically replans vehicle waypoints in real time based on image coverage of an area I have selected to map. 
I don't want to autoload this module because it will only run on higher spec hardware but I also want to ship it with maverick api for those who want to use it. How would this be achieved using the system you described?
Your point of having a separate location for 3rd party modules is a good one. I was thinking that -api could look in `~/code/api-modules` or similar at runtime to discover external modules

SamuelDudley, 2018-12-20:
Another example is a module that only works for a rover, plane or copter. If I have different vehicle types I personally don't want to delete or move module code around.

fnoop, 2018-12-20:
Oh I must have misunderstood - where the modules you had in modules/ already, not autoloaded?  I agree defining them in config is better.

fnoop, 2018-12-22:
Rename modules/graphql to modules/api

closed 2018-12-22 by fnoop