goodrobots/maverick-api #54: Document API

fnoop, 2018-12-24:
Need to document the gql api for endpoints to use (ie what messages and schema are available, field types etc).
Possible to autogenerate the docs?

SamuelDudley, 2018-12-27:
Some options

https://github.com/gjtorikian/graphql-docs/blob/master/README.md

https://github.com/2fd/graphdoc

Doesn't seem to be a python version out there yet.

It would be great to go one step further and show what endpoints require auth and at what level.

SamuelDudley, 2018-12-28:
Looks like drone deploy is using https://github.com/2fd/graphdoc#readme

SamuelDudley, 2020-02-05:
docs are now being built as a github action https://github.com/goodrobots/maverick-api/commit/ede434f186281fc02579a4b45e3197ad62a65c4d

https://goodrobots.github.io/maverick-api/

Note that the docs are not great... need to look into other options. Seems like drone deploy has a private fork of https://github.com/2fd/graphdoc and is generating the documentation from there (much nicer).

For now its better than nothing...

closed 2020-02-05 by SamuelDudley

fnoop, 2020-02-05:
Looks good!  Is all the mavros related stuff missing because the modules are disabled?

SamuelDudley, 2020-02-05:
Correct. The schema is built as the modules are loaded and as mavros is disabled the schema is missing.

fnoop, 2020-02-05:
Can you build the schema for all modules instead?  I would have thought for public docs we would want all possibly modules documented?

SamuelDudley, 2020-02-05:
@fnoop see https://github.com/goodrobots/maverick-api/issues/208