goodrobots/maverick-api #67: Rename old ...Message message names

fnoop, 2018-12-26:
@SamuelDudley I think we agreed to rename the old message names, like StateMessage to State?  Am I remembering that correctly?

SamuelDudley, 2018-12-26:
Happy to do so but I think we should namespace the messages with the module name. Otherwise we might clash when adding mavlink messages, etc... Thoughts?

fnoop, 2018-12-26:
Wouldn't mavlink be a different provider to the same graphql messages?

SamuelDudley, 2018-12-26:
If that's the case, we need to change the current messages as they reflect mavros pretty much directly

fnoop, 2018-12-26:
Would we want to provide the same message from different sources?  I would have thought mavros/mavlink would just be different backends for the same graphql messages, so the frontend is unaware of the backend type?

SamuelDudley, 2018-12-26:
That makes total sense from a ui perspective, rather than supporting two messages in the client that do the same thing. Currently I have reflected the mavros messages but I think we need to capture our needs for each message and send them instead. I'm also happy to reflect mavros in addition to our custom messages even if data is duplicated. 
Happy to work with what we have for the moment, but making the messages more targeted for our application would be beneficial imo.

SamuelDudley, 2019-01-02:
name changes applied in https://github.com/goodrobots/maverick-api/commit/cd20dca66c1793c8ba3b4695366c58d1774a5825.
closing this until we have another backend to push similar data to -web. We can review the message content and names at that time

closed 2019-01-02 by SamuelDudley