goodrobots/maverick-api #71: Too many logfiles

fnoop, 2019-01-03:
New logging is now split into too many logfiles.  example log directory for single instance:
```
[dev] [mav@goodrobots ~/var/log/web/api/apsitl_dev]$ ls -lrt
total 3192
-rw-r--r-- 1 mav mav     414 Jan  2 10:52 general.log
-rw-r--r-- 1 mav mav 1048419 Jan  2 23:25 websocket.log.2
-rw-r--r-- 1 mav mav    8436 Jan  2 23:48 root.log
-rw-r--r-- 1 mav mav 1048419 Jan  3 00:25 websocket.log.1
-rw-r--r-- 1 mav mav   19928 Jan  3 07:30 access.log
-rw-r--r-- 1 mav mav  232402 Jan  3 07:30 log
-rw-r--r-- 1 mav mav  467850 Jan  3 07:30 api.log
-rw-r--r-- 1 mav mav  413603 Jan  3 07:38 websocket.log
```

 - Create archive directory for rotated logs
 - Reduce number of logfiles, too confusing to find what you need.  Current log files are:
```
general.log
root.log
access.log
log
api.log
websocket.log
```

SamuelDudley, 2019-01-03:
`websocket.log` only gets written to when in debug as it is too verbose within `api.log`.

That leaves:
 - `general.log` which contains only `tornado.general` logging events. This could probably be rolled into `api.log`
 - `access.log` which contains only `tornado.access` logging events. I feel these should be separate for audit purposes.
 - `api.log` which contains `tornado.application` and eventually all of the `base` modules (this is still a work in progress) 
 - I created `root.log` to get rid of `log` (this existed before) as things like mavros need to log somewhere... I can see that that has failed but that is the intention.

I actually wanted to add more logs! (to keep the non-base modules out of the `api.log`).

Re moving the files, this will be tricky as its not a standard feature of the rotating file handler, as a log gets too big it is pushed from `foo.log` into `foo.log.1`, which pushes `foo.log.1` into `foo.log.2`  and so on with the content of `foo.log.2` being dropped forever.

TL;DR the logs without suffix numbers are the current active logs.

SamuelDudley, 2019-01-07:
My understanding is that as the root logger isn't handled, things such as mavros logging calls will end up in `log`.  I believe that as `api.log` is a child of the root logger its messages are being propagated up to `log` after being handled in `api.log`.
I'll have a go at setting `propagate` to `False` for the loggers that we write to disc (such as `api.log`). This should solve the duplication issue.

SamuelDudley, 2019-01-07:
attempted the above. Turns out the content of `log` is from the following segment of code and only gets created when api is run as a service:

```
channel = logging.StreamHandler()
            channel.setFormatter(
            LogFormatter(
                color=True,
                fmt="%(color)s" + self.log_fmt_prefix + "%(end_color)s%(message)s",
                datefmt=self.date_fmt,
                colors={10: 4, 20: 2, 30: 3, 40: 1},
                )
            )
            self.application_log.addHandler(channel)
```
We can add a config option to avoid running this code (and duplicating the log data) but currently the `log` file is being created regardless of the `StreamHandler` existing.

SamuelDudley, 2019-01-13:
@fnoop what are your thoughts on the `log` file and if we want it in that directory?

fnoop, 2019-01-21:
Got rid of api.sh redirect to `log` file - instead just leave the output to stdout which gets captured by system journal when run from systemd.  (In maverick therefore the output can be viewed with eg. `maverick log api@apsitl_dev`, or on non-maverick systems `sudo journalctl -u api@apsitl_dev`

https://github.com/goodrobots/maverick/commit/66f00739a7c3ef37916b5b3277f935ed657106fc

closed 2019-01-21 by fnoop