goodrobots/maverick-api #74: Expose firmware and vehicle types in graphql

fnoop, 2019-01-09:
Graphql should show the firmware type (px4, ardupilot etc), and the vehicle type (copter, plane etc)

SamuelDudley, 2019-01-09:
info returned from mavros (per connected vehicle):
```
    success: True
    vehicles: 
      - 
        header: 
          seq: 0
          stamp: 
            secs: 1547036032
            nsecs: 255436314
          frame_id: ''
        available_info: 3
        sysid: 1
        compid: 1
        autopilot: 3
        type: 2
        system_status: 4
        base_mode: 217
        custom_mode: 17
        mode: "BRAKE"
        mode_id: 17
        capabilities: 15311
        flight_sw_version: 50790400
        middleware_sw_version: 0
        os_sw_version: 0
        board_version: 0
        vendor_id: 0
        product_id: 0
        uid: 0
```

`autopilot` field is then being parsed to return the frame type e.g. `ArduCopter`

SamuelDudley, 2019-01-13:
First slab of work completed with https://github.com/goodrobots/maverick-api/commit/39593f1536c79705733cd213ab19d54bb72cad08

only runs one at API startup. Need to setup a 1Hz callback to looks for additional system(s) that come online

fnoop, 2019-01-13:
This is perfect for initial requirements!

fnoop, 2019-01-13:
re: only running once - I looked at the data and thought that we only need it once, so I'm only using an initial query to grab the data and not even bothering with a subscription.
No doubt we'll come across use cases for updating the data in the future, but for now I think we're good.