goodrobots/maverick-api #75: Query parameters not working

fnoop, 2019-01-11:
Not sure where the problem is - could well be the client - but I think it's formulating the query correctly.  api.log shows:

```
[D 2019-01-11,07:35:17 /srv/maverick/code/maverick-api/maverick_api/modules/base/tornadoql/graphql_handler.py:95]
    GraphQL result data: None errors: [GraphQLError("Variable '$id' of required type 'String!' was not provided.", locations=[SourceLocation(line=1, column=24)])]
[W 2019-01-11,07:35:17 /srv/maverick/code/maverick-api/maverick_api/modules/base/tornadoql/graphql_handler.py:101]
    GraphQL Error: Variable '$id' of required type 'String!' was not provided.

    GraphQL request (1:24)
    1: query missionListQuery($id: String!) {
                              ^
    2:   MissionList(id: $id) {
```

Chrome shows the query variables being sent:
[image]

SamuelDudley, 2019-01-11:
Most likely API side... I dont think im passing the `variables` field into the query. I'll have a look.

SamuelDudley, 2019-01-11:
Fixed by https://github.com/goodrobots/maverick-api/commit/7dc7529a4ce7a2d60f9d52ff1d178f1131f9d7a2

@fnoop Can you confirm?

fnoop, 2019-01-11:
Fixed, thanks!

closed 2019-01-11 by fnoop