goodrobots/maverick-api #87: Convert all error output to client-friendly json

fnoop, 2019-01-22:
Currently for most of the errors in -api, it returns a string eg. 
```
[W 2019-01-22,13:38:32 /srv/maverick/code/maverick-api/maverick_api/modules/base/tornadoql/graphql_handler.py:101]
    GraphQL Error: [Errno 2] No such file or directory: '/srv/maverick/data/web/api/apsitl_custom/missions/test.mission'

    GraphQL request (2:3)
    1: query missionListQuery($id: String!) {
    2:   MissionList(id: $id) {
         ^
    3:     ...MissionList
```

The browser apollo library can't interpret this, so the dev console just shows:
`Error Network error: Unexpected token G in JSON at position 0`

It shows the same message for every error.  It would be very helpful to have the actual error returned in the browser, so we don't have to go hunting in the api logs for every error.

SamuelDudley, 2019-01-22:
If you can provide a json format which will behave with apollo, I'll implement it :)

fnoop, 2019-01-22:
You fixed this before for one error, can't remember which one it was - so it displayed correctly in the client :)

SamuelDudley, 2019-01-22:
I fixed the websocket errors, but the query data was more temperamental from memory. I'll wrap the string up in some json and see what apollo thinks of it.

SamuelDudley, 2019-01-23:
goodrobots/maverick-api@de7d826
@fnoop See my comment on gitter and let me know if you would like anything done to the JSON structure

SamuelDudley, 2019-03-11:
Looks like the latest release will help with outputting more useful error messages: https://github.com/graphql-python/graphql-core-next/issues/23

SamuelDudley, 2020-02-05:
traceback added https://github.com/goodrobots/maverick-api/commit/4a4aeff802599618695d68d05b6fa471b0361b76

closed 2020-02-05 by SamuelDudley