goodrobots/maverick-api #121: Track graphql-core developments

fnoop, 2019-03-25:
Since we moved to graphql-core-next and python3, the roadmap for graphene has changed to incorporate the work from graphql-core-next back into a 'modern' branch of graphql-core:

https://github.com/graphql-python/graphene/issues/833#issuecomment-427574227

It's probably worth waiting until that work is complete, as it looks like graphql-core-next is still being actively developed.

SamuelDudley, 2019-08-03:
An update on this. Graphql-core has been updated to use the latest revision of graphql-core-next.
Graphql-core-next is no longer being developed.

Basically we need to move from graphql-core-next to graphql-core>=3

SamuelDudley, 2019-09-10:
everything I said above is wrong... Graphql-core-next is the package to use going forward. No change required. Closing this issue as Graphql-core-next has stabilized.

SamuelDudley, 2019-09-16:
so although the github repo has changed to graphql-core-next the pip name has not...
install the latest (published) version with `pip install graphql-core==3.0.0b0`
install the latest (master) version with `pip install git+https://github.com/graphql-python/graphql-core-next`

SamuelDudley, 2020-02-04:
Updated requirements to use latest version of graphql-core https://github.com/goodrobots/maverick-api/commit/707633a4d93d338d370d9018a8f4b0c873a656e6

closed 2020-02-04 by SamuelDudley