goodrobots/maverick-api #122: Missing requirements

fnoop, 2019-03-25:
```
Mar 25 12:34:04 goodrobots api.sh[13002]:     import jstyleson as jjson
Mar 25 12:34:04 goodrobots api.sh[13002]: ModuleNotFoundError: No module named 'jstyleson'
```

fnoop, 2019-03-25:
```
Mar 25 14:01:57 goodrobots api.sh[17745]:     from jsondiff import diff as jdiff
Mar 25 14:01:57 goodrobots api.sh[17745]: ModuleNotFoundError: No module named 'jsondiff'
```

fnoop, 2019-03-25:
maverick currently keeps a separate type of pip install for -api requirements, eg:
```
# install python components
    install_python_module { "mavapi-pymavlink":
        pkgname     => "pymavlink",
        ensure      => atleast,
        version     => "2.2.10",
    } ->
    install_python_module { "mavapi-graphql-core-next":
        pkgname     => "GraphQL-core-next",
        ensure      => exactly,
        version     => "1.0.2",
    }
    install_python_module { "mavapi-tornado":
        pkgname     => "tornado",
        ensure      => exactly,
        version     => "6.0.2",
    } ->
    install_python_module { "mavapi-aiosqlite":
        pkgname     => "aiosqlite",
        ensure      => exactly,
        version     => "0.8.1",
    } ->
    install_python_module { "mavapi-jstyleson":
        pkgname     => "jstyleson",
        ensure      => exactly,
        version     => "0.0.2",
    } ->
```

Instead, use the requirements.txt from -api repo.

fnoop, 2019-03-25:
```
Mar 25 14:20:01 goodrobots api.sh[1140]:     import yaml
Mar 25 14:20:01 goodrobots api.sh[1140]: ModuleNotFoundError: No module named 'yaml'
```

fnoop, 2019-03-25:
```
Mar 25 14:22:07 goodrobots api.sh[1642]:     import rospkg
Mar 25 14:22:07 goodrobots api.sh[1642]: ModuleNotFoundError: No module named 'rospkg'
```

fnoop, 2019-03-25:
Adding these packages with pip3 gets us going again:
```
jstyleson==0.0.2
jsondiff==1.1.2
pyyaml==5.1
rospkg==1.1.7
```

SamuelDudley, 2019-03-29:
So the general idea was to use pipenv to manage requirements https://pipenv.readthedocs.io/en/latest/
requirements.txt and requirements-dev.txt is generated from pipenv. There are other options such as poetry https://poetry.eustace.io/docs/

From what I understand pipenv was the defacto standard, but that may have changed.
I'll look into it a bit more.

SamuelDudley, 2019-03-29:
Looks like both have issues with poetry having less issues in general...