goodrobots/maverick-api #204: Info Invalid GraphQL 'VehicleInfo' data returned from api: dev

fnoop, 2020-02-02:
This happens quite a bit.  Not sure where the error is - api or web (requesting the wrong thing), but it should be a clearer error.  The error itself comes from -api, so -api should handle errors better and present a better message to the frontend, and to it's own logs.

SamuelDudley, 2020-02-02:
[image]

The issue was / is that the python graphql library swallows the entire stack trace leaving -api with only the error message you see in the console. 

related info in #87

SamuelDudley, 2020-02-05:
-api traceback added to both -api logs and the web front end via json https://github.com/goodrobots/maverick-api/commit/4a4aeff802599618695d68d05b6fa471b0361b76

fnoop, 2020-03-03:
As per #39, vehicleinfo is missing from mavros topics, so it mustn't have been folded into mavros.

However -api still adds it to the available schema even if it isn't available, and so -web verifies it true and then tries to use it.

fnoop, 2020-03-03:
Ah, we need to migrate to what vehicleinfo turned into:
https://github.com/mavlink/mavros/pull/1150

closed 2020-03-04 by fnoop