goodrobots/maverick-api #213: Add zeroconf broadcast of -api

fnoop, 2020-02-29:
-api should announce itself on the network so clients (-web) can find it.

SamuelDudley, 2020-02-29:
https://github.com/jstasiak/python-zeroconf/blob/master/examples/registration.py

SamuelDudley, 2020-03-01:
@fnoop maverick-api now looks for other api and announces itself via zeroconf

SamuelDudley, 2020-03-02:
Use `socket.getfqdn()`

fnoop, 2020-03-02:
clashes with avahi:
```
[I 2020-03-02,10:01:13 /srv/maverick/code/maverick-api/maverick_api/modules/__init__.py:112]
    Loading module: api.status
Traceback (most recent call last):
  File "/srv/maverick/code/maverick-api/maverick-api.py", line 41, in <module>
    api.initialize()
  File "/srv/maverick/code/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 68, in initialize
    self.discovery_zeroconf_module = DiscoveryZeroconfModule(loop, module_schema)
  File "/srv/maverick/code/maverick-api/maverick_api/modules/api/discovery/discovery_zeroconf.py", line 34, in __init__
    self.zeroconf = Zeroconf(ip_version=ip_version)
  File "/srv/maverick/software/python/lib/python3.7/site-packages/zeroconf/__init__.py", line 2134, in __init__
    interfaces, unicast, ip_version, apple_p2p=apple_p2p
  File "/srv/maverick/software/python/lib/python3.7/site-packages/zeroconf/__init__.py", line 2054, in create_sockets
    listen_socket = new_socket(ip_version=ip_version, apple_p2p=apple_p2p)
  File "/srv/maverick/software/python/lib/python3.7/site-packages/zeroconf/__init__.py", line 1992, in new_socket
    s.bind(('', port))
OSError: [Errno 98] Address already in use
```

fnoop, 2020-03-02:
Think we need to check if a zeroconf is already running or present on the network, and if so then register with it

fnoop, 2020-03-02:
Or it's possible we can start on a different port and co-exist

SamuelDudley, 2020-03-02:
Darn...

fnoop, 2020-03-04:
Possible fix for clashing zeroconf stacks:
https://github.com/goodrobots/maverick/issues/912

Also note that -api is advertising with a single generic service name:
`+   eth0 IPv4 Maverick API                                  Web Site             local`
(see `avahi-browse -a`)
I think this means that -api running on multiple devices will clash.

SamuelDudley, 2020-03-05:
Updated the name to include the -api uuid. Hopefully no clashes now :)

SamuelDudley, 2020-03-16:
Need to add service browser support so that other -api instances and visiond can be found and sent via graphql to -web
https://github.com/jstasiak/python-zeroconf/blob/master/examples/browser.py

fnoop, 2020-03-20:
Could you add -api ssl state to the broadcast?

fnoop, 2020-03-20:
Could you please also add the unique id?

SamuelDudley, 2020-03-20:
https://github.com/goodrobots/maverick-api/blob/master/maverick_api/modules/api/discovery/discovery_zeroconf.py#L32

Depending on if SSL is used the broadcast should already give the endpoints as http/https. The unique ID is in the broadcast name, do you need an line in properties with that also?

fnoop, 2020-03-20:
I've added uuid and service_type fields to properties dict - is the uuid unique to the instance of -api?  If there are multiple -api running on the same computer will they get different uuids, and is it the same after -api is restarted?

SamuelDudley, 2020-03-20:
It will be unique to the server, but it currently resets to a new value every boot. I need to make the uuid persistent in the -api database.

fnoop, 2020-03-20:
Sorry when you say unique to the server, does that mean the computer server or the -api server?

Yes please to persistent - we'll need that to recognise it in the -web over time, otherwise it will keep trying to create new clients.

SamuelDudley, 2020-03-20:
Individual ID per API instance. The plan was always

fnoop, 2020-03-29:
uuid4 is not creating repeatable ids, so everytime the machine is rebooted -web discovers duplicate api connections.  This doesn't seem to happen with visiond, so uuid5 looks good.

closed 2020-03-30 by SamuelDudley