goodrobots/maverick-api #219: mavros modules broken

fnoop, 2020-03-05:
After recent module refactor, mavros modules are not loading again - they were working fine before the refactor:
https://github.com/goodrobots/maverick-api/commit/f3673590859fdb41a72180516708514c97d16163

```
Mar 05 16:23:44 dev.maverick.one api.sh[25397]: [I 2020-03-05,16:23:44 /srv/maverick/code/maverick-api/maverick_api/modules/__init__.py:135]
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:     Appending schema from maverick_api.modules.api.mavros.ImuSchema
Mar 05 16:23:44 dev.maverick.one api.sh[25397]: Traceback (most recent call last):
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:   File "/srv/maverick/code/maverick-api/maverick-api.py", line 41, in <module>
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:     api.initialize()
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:   File "/srv/maverick/code/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 46, in initialize
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:     generate_schema()
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:   File "/srv/maverick/code/maverick-api/maverick_api/modules/__init__.py", line 136, in generate_schema
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:     _module_schema[ref_name] = attribute()
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:   File "/srv/maverick/code/maverick-api/maverick_api/modules/api/mavros/mavros_imu.py", line 23, in __init__
Mar 05 16:23:44 dev.maverick.one api.sh[25397]:     super().__init__()
Mar 05 16:23:44 dev.maverick.one api.sh[25397]: TypeError: __init__() missing 1 required positional argument: 'child'
```

fnoop, 2020-03-05:
Added 'self' to super calls which seems to have fixed those errors.  Now getting:
```
Mar 05 16:26:53 dev.maverick.one api.sh[28438]: [I 2020-03-05,16:26:53 /srv/maverick/code/maverick-api/maverick_api/modules/__init__.py:145]
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:     Creating module instance: maverick_api.modules.api.mavros.MAVROSConnection
Mar 05 16:26:53 dev.maverick.one api.sh[28438]: Traceback (most recent call last):
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:   File "/srv/maverick/code/maverick-api/maverick-api.py", line 41, in <module>
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:     api.initialize()
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:   File "/srv/maverick/code/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 46, in initialize
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:     generate_schema()
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:   File "/srv/maverick/code/maverick-api/maverick_api/modules/__init__.py", line 146, in generate_schema
Mar 05 16:26:53 dev.maverick.one api.sh[28438]:     _module_base[ref_name] = attribute()
Mar 05 16:26:53 dev.maverick.one api.sh[28438]: TypeError: __init__() missing 2 required positional arguments: 'loop' and 'module'
```
aargh

fnoop, 2020-03-05:
@SamuelDudley dev node has a working sitl (api@dev service)

SamuelDudley, 2020-03-06:
Reported on gitter as resolved

closed 2020-03-06 by SamuelDudley