goodrobots/maverick-api #223: add human name to configuration and serve via zeroconf

SamuelDudley, 2020-03-22:
The name will be defined in config and set via maverick when a "service group" is created e.g. dev or sitl

The name can be overwritten / set to whatever the user wants but if created by maverick will be set to the name defined for the "service group". Ideally the user has the ability to set the name in localconf.

fnoop, 2020-03-24:
As well as a name for the vehicle, would be great to also have a cluster/swarm/group name as well.  eg.
Name=DevFrame 350 v1
Group=Garage Swarm

fnoop, 2020-03-26:
Added a 'Name' parameter to the config which will be shown if set:
`+  wman0 IPv4 maverick-api (Maverick)                       _api._tcp            local`

 If not, it will generate a name from hostname:port, for uniqueness:
`+  wman0 IPv4 maverick-api (maverick-nano:7000)             _api._tcp            local`

fnoop, 2020-03-29:
Added in #224

closed 2020-03-29 by fnoop