goodrobots/maverick-api #228: Low-level websocket error when connecting to -api

fnoop, 2020-04-08:
Looks like when initial subscription is made the websocket gets cut off or interrupted.  Is it from the server side?  Can't see anything obvious in the logs.
[image]

[image]

SamuelDudley, 2020-04-08:
The top one is new to me. Do you think it has something to do with the client attempting to connect to /subscriptions before the schema has loaded? Will investigate further on -api side.

I have certainly seen the bottom one before. I think it's related to the browser trying to reconnect to a websocket -api has closed. Possibly -api is closing them incorrectly. I'll investigate.

fnoop, 2020-04-08:
Sorry should have been clearer - they're both the same thing but seen in different browsers.  The top one is firefox, the bottom one is chrome.  This happens when the page is first loaded, and the clients are first loaded.  It's happening right down at the apollo client level.
We've changed all the client creation to async, but it doesn't try to create the initial query/subscription until the schema has been fully downloaded and parsed:
```
          // If schema has been fetched, proceed to create Status query/subscription
          if (newValue[api].schemaready === true) {
            this.createQuery('Status', statusQuery, api, null, null, this.processStatusQuery)
            this.createSubscription('Status', statusSubscription, api, null, null, this.processStatusSubscription)
          }
```
It's entirely possible it's something to do with timing at the client end..

SamuelDudley, 2020-04-08:
Okay. If you close the web browser, restart -api and then load the page do you see the error?

fnoop, 2020-04-08:
No, although actually if I hit reload lots of times it happens sometimes and not others.  Odd.  I guess that probably points to a brower-side timing issue with the async calls?

SamuelDudley, 2020-04-08:
I actually think it's to do with -api and websocket connections not being disposed of correctly. There is an error that occurs in the logs I see every now and then generated from here https://github.com/goodrobots/maverick-api/blob/master/maverick_api/modules/base/tornadoql/subscription_handler.py#L211