goodrobots/maverick-api #257: Error starting up

fnoop, 2020-08-05:
```
Traceback (most recent call last):
  File "/srv/maverick/software/maverick-api/maverick-api.py", line 40, in <module>
    api.initialize()
  File "/srv/maverick/software/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 47, in initialize
    generate_schema()
  File "/srv/maverick/software/maverick-api/maverick_api/modules/__init__.py", line 129, in generate_schema
    f"{__name__}.{module_folder_name}.{module_name}"
  File "/srv/maverick/software/python/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/__init__.py", line 2, in <module>
    from maverick_api.modules.api.mavros.mavros_connection import MAVROSConnection
  File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/mavros_connection.py", line 21, in <module>
    from maverick_api.modules.api.mavros.mavros_param import ParamInterface
  File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/mavros_param.py", line 42, in <module>
    from graphql.error import INVALID
ImportError: cannot import name 'INVALID' from 'graphql.error' (/srv/maverick/software/python/lib/python3.7/site-packages/graphql/error/__init__.py)
```

@SamuelDudley - maybe something to do with a recent graphql update?

SamuelDudley, 2020-08-05:
I'll check it out. Current master everything?

fnoop, 2020-08-06:
yes just double checked to make sure :)

SamuelDudley, 2020-08-17:
I'm struggling to get a working environment to reproduce. Using the latest beta on a pi3b+ rosmaster fails to start:
```
[roslaunch][INFO] 2020-08-17 09:44:54,887: Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2020-08-17 09:44:54,898: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2020-08-17 09:44:54,899: roslaunch starting with args ['roscore', '--core', '-p', '6114']
[roslaunch][INFO] 2020-08-17 09:44:54,900: roslaunch env is environ({'SHELL': '/bin/bash', 'ROS_VERSION': '1', 'PKG_CONFIG_PATH': '/srv/maverick/software/ros/current/lib/pkgconfig:/opt/ros/current/lib/pkgconfig:/srv/maverick/software/ros/noetic/lib/pkgconfig', 'ROS_PYTHON_VERSION': '3', 'ROS_PACKAGE_PATH': '/srv/maverick/software/ros/current/share:/opt/ros/current/share:/srv/maverick/software/ros/noetic/share', 'ROSLISP_PACKAGE_DIRECTORIES': '', 'PWD': '/', 'LOGNAME': 'mav', 'ROS_LOG_DIR': '/srv/maverick/var/log/ros/apdev', 'HOME': '/srv/maverick', 'LANG': 'en_GB.UTF-8', 'ROS_ETC_DIR': '/srv/maverick/software/ros/current/etc/ros', 'CMAKE_PREFIX_PATH': '/srv/maverick/software/ros/current:/opt/ros/current:/srv/maverick/software/ros/noetic', 'INVOCATION_ID': '697549e52c1d4199b31d47220bd996e5', 'PYTHONPATH': '/srv/maverick/software/ros/current/lib/python2.7/dist-packages:/srv/maverick/software/ros/current/lib/python3/dist-packages:/srv/maverick/software/ros/current/lib/python3.7/site-packages:/srv/maverick/software/ros/current/lib/python3.7/site-packages:/opt/ros/current/lib/python3.7/site-packages:/srv/maverick/software/ros/noetic/lib/python3.7/site-packages', 'USER': 'mav', 'ROS_HOME': '/srv/maverick/var/log/ros/apdev', 'SHLVL': '1', 'ROS_MASTER_URI': 'http://localhost:6114', 'LD_LIBRARY_PATH': '/srv/maverick/software/ros/current/lib:/opt/ros/current/lib:/srv/maverick/software/ros/noetic/lib', 'ROS_PORT': '6114', 'JOURNAL_STREAM': '8:40216', 'PATH': '/srv/maverick/software/ros/current/bin:/opt/ros/current/bin:/srv/maverick/software/ros/noetic/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin', 'ROS_ROOT': '/srv/maverick/software/ros/current/share/ros', 'ROS_DISTRO': 'noetic', '_': '/srv/maverick/software/python/bin/python3', 'ROS_LOG_FILENAME': '/srv/maverick/var/log/ros/apdev/ec18abcc-e065-11ea-80d3-b827eb2b499a/roslaunch-maverick-raspberry-3955.log'})
[roslaunch][INFO] 2020-08-17 09:44:54,901: starting in server mode
[roslaunch.parent][INFO] 2020-08-17 09:44:54,902: starting roslaunch parent run
[roslaunch][INFO] 2020-08-17 09:44:54,903: loading roscore config file /srv/maverick/software/ros/current/etc/ros/roscore.xml
[roslaunch][ERROR] 2020-08-17 09:44:54,919: RLException: Invalid <param> tag: Cannot load command parameter [rosversion]: no such command [['rosversion', 'roslaunch']]. 

Param xml is <param command="rosversion roslaunch" name="rosversion"/>
[roslaunch][ERROR] 2020-08-17 09:44:54,919: The traceback for the exception was written to the log file
[roslaunch][ERROR] 2020-08-17 09:44:54,928: Traceback (most recent call last):
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/loader.py", line 529, in param_value
    p = subprocess.Popen(command, stdout=subprocess.PIPE)
  File "/srv/maverick/software/python/lib/python3.7/subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "/srv/maverick/software/python/lib/python3.7/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'rosversion': 'rosversion'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/xmlloader.py", line 271, in _param_tag
    value = self.param_value(verbose, name, ptype, *vals)
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/loader.py", line 537, in param_value
    raise ValueError("Cannot load command parameter [%s]: no such command [%s]"%(name, command))
ValueError: Cannot load command parameter [rosversion]: no such command [['rosversion', 'roslaunch']]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/__init__.py", line 347, in main
    p.start()
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/parent.py", line 305, in start
    self._start_infrastructure()
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/parent.py", line 254, in _start_infrastructure
    self._load_config()
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/parent.py", line 157, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/config.py", line 451, in load_config_default
    load_roscore(loader, config, verbose=verbose)
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/config.py", line 94, in load_roscore
    loader.load(f_roscore, config, core=True, verbose=verbose)
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/xmlloader.py", line 761, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
[... cut ...]
    return f(*args, **kwds)
  File "/srv/maverick/software/ros/current/lib/python3.7/site-packages/roslaunch/xmlloader.py", line 286, in _param_tag
    "Invalid <param> tag: %s. \n\nParam xml is %s"%(e, tag.toxml()))
roslaunch.xmlloader.XmlParseException: Invalid <param> tag: Cannot load command parameter [rosversion]: no such command [['rosversion', 'roslaunch']]. 

Param xml is <param command="rosversion roslaunch" name="rosversion"/>

[rospy.core][INFO] 2020-08-17 09:44:54,931: signal_shutdown [atexit]
```

fnoop, 2020-08-17:
```
[dev] [mav@maverick-raspberry ~]$ ~/software/maverick/bin/rosmaster.sh apdev
... logging to /srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/roslaunch-maverick-raspberry-7991.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://maverick-raspberry:45979/
ros_comm version 1.15.8

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES

auto-starting new master
process[master]: started with pid [8001]
ROS_MASTER_URI=http://maverick-raspberry:6114/

setting /run_id to 69809628-e067-11ea-82a7-dca6320a583b
process[rosout-1]: started with pid [8012]
started core service [/rosout]
/srv/maverick/software/ros/current/lib/rosout/rosout: error while loading shared libraries: libboost_filesystem.so.1.67.0: cannot open shared object file: No such file or directory
[rosout-1] process has died [pid 8012, exit code 127, cmd /srv/maverick/software/ros/current/lib/rosout/rosout __name:=rosout __log:=/srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1.log].
log file: /srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1*.log
[rosout-1] restarting process
process[rosout-1]: started with pid [8015]
/srv/maverick/software/ros/current/lib/rosout/rosout: error while loading shared libraries: libboost_filesystem.so.1.67.0: cannot open shared object file: No such file or directory
[rosout-1] process has died [pid 8015, exit code 127, cmd /srv/maverick/software/ros/current/lib/rosout/rosout __name:=rosout __log:=/srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1.log].
log file: /srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1*.log
[rosout-1] restarting process
process[rosout-1]: started with pid [8016]
/srv/maverick/software/ros/current/lib/rosout/rosout: error while loading shared libraries: libboost_filesystem.so.1.67.0: cannot open shared object file: No such file or directory
[rosout-1] process has died [pid 8016, exit code 127, cmd /srv/maverick/software/ros/current/lib/rosout/rosout __name:=rosout __log:=/srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1.log].
log file: /srv/maverick/var/log/ros/apdev/69809628-e067-11ea-82a7-dca6320a583b/rosout-1*.log
```

can't find boost libraries..

SamuelDudley, 2020-08-17:
I know `/srv/maverick/software/maverick/bin/api.sh` does some setup of the python path re ROS but im a buit stuck on how to begin debugging the issue further. Happy to try out ideas if you have any?

SamuelDudley, 2020-08-17:
I have deeper issues:
```
mav@maverick-raspberry:~$ ~/software/maverick/bin/rosmaster.sh apdev
... logging to /srv/maverick/var/log/ros/apdev/ead9615a-e067-11ea-b64a-b827eb2b499a/roslaunch-maverick-raspberry-5375.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Unable to contact my own server at [http://maverick-raspberry:34017/].
This usually means that the network is not configured properly.

A common cause is that the machine cannot ping itself.  Please check
for errors by running:

        ping maverick-raspberry

For more tips, please see

        http://wiki.ros.org/ROS/NetworkSetup

The traceback for the exception was written to the log file
mav@maverick-raspberry:~$ ping maverick-raspberry
PING maverick-raspberry.local (192.168.0.12) 56(84) bytes of data.
From 192.168.0.11 (192.168.0.11) icmp_seq=1 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=2 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=3 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=4 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=5 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=6 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=7 Destination Host Unreachable
From 192.168.0.11 (192.168.0.11) icmp_seq=8 Destination Host Unreachable
^C
--- maverick-raspberry.local ping statistics ---
8 packets transmitted, 0 received, +8 errors, 100% packet loss, time 17ms
pipe 4
mav@maverick-raspberry:~$ ifconfg
bash: ifconfg: command not found
mav@maverick-raspberry:~$ ip
ip                           ip6tables-legacy-save        ip6tables-restore-translate  ipcs                         iptables-legacy-restore      iptables-restore             iptunnel
ip6tables                    ip6tables-nft                ip6tables-save               ipmaddr                      iptables-legacy-save         iptables-restore-translate   
ip6tables-apply              ip6tables-nft-restore        ip6tables-translate          iptables                     iptables-nft                 iptables-save                
ip6tables-legacy             ip6tables-nft-save           ipcmk                        iptables-apply               iptables-nft-restore         iptables-translate           
ip6tables-legacy-restore     ip6tables-restore            ipcrm                        iptables-legacy              iptables-nft-save            iptables-xml                 
mav@maverick-raspberry:~$ ifconfig 
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:7e:1c:cf  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2014530  bytes 901134647 (859.3 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2014530  bytes 901134647 (859.3 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.11  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 2001:8003:1848:8f01:ba27:ebff:fe2b:499a  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::ba27:ebff:fe2b:499a  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:2b:49:9a  txqueuelen 1000  (Ethernet)
        RX packets 26137  bytes 30980774 (29.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16158  bytes 3738666 (3.5 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

[... 38 lines cut ...]
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2062583  bytes 922520567 (879.7 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2062583  bytes 922520567 (879.7 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.11  netmask 255.255.255.0  broadcast 192.168.0.255
        inet6 2001:8003:1848:8f01:ba27:ebff:fe2b:499a  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::ba27:ebff:fe2b:499a  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:2b:49:9a  txqueuelen 1000  (Ethernet)
        RX packets 26321  bytes 31012955 (29.5 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 16379  bytes 3776891 (3.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mav@maverick-raspberry:~$ 
```
Appears to be a hostname issue?

fnoop, 2020-08-17:
> Appears to be a hostname issue?

Looks like your dhcp lease has renewed with a different IP.  Either change the entry in /etc/hosts with the new IP or run `maverick configure`, which should it for you.  The networking will be completely revamped in the next version, I thought I'd try and get this problem child out the door first!

SamuelDudley, 2020-08-17:
okay, yep. /etc/hosts was pointing to the wrong ip. Running a configure now.

SamuelDudley, 2020-08-17:
okay. I'm now seeing the same boost issue as you.

SamuelDudley, 2020-08-17:
```
mav@maverick-raspberry:/usr/local/lib$ sudo ldconfig -v | grep boost
ldconfig: Can't stat /usr/local/lib/arm-linux-gnueabihf: No such file or directory
ldconfig: Path `/lib/arm-linux-gnueabihf' given more than once
ldconfig: Path `/usr/lib/arm-linux-gnueabihf' given more than once
ldconfig: /lib/arm-linux-gnueabihf/ld-2.28.so is the dynamic linker, ignoring

        libboost_program_options.so.1.67.0 -> libboost_program_options.so.1.67.0
        libboost_system.so.1.67.0 -> libboost_system.so.1.67.0
        libboost_thread.so.1.67.0 -> libboost_thread.so.1.67.0
        libboost_iostreams.so.1.58.0 -> libboost_iostreams.so.1.58.0
        libboost_regex.so.1.67.0 -> libboost_regex.so.1.67.0
        libboost_iostreams.so.1.67.0 -> libboost_iostreams.so.1.67.0
        libboost_atomic.so.1.67.0 -> libboost_atomic.so.1.67.0
ldconfig: /lib/ld-linux.so.3 is the dynamic linker, ignoring

mav@maverick-raspberry:/usr/local/lib$ 
```

SamuelDudley, 2020-08-17:
`sudo apt-get install libboost1.67-all -y`

```
mav@maverick-raspberry:/usr/local/lib$ sudo ldconfig -v | grep boost
ldconfig: Can't stat /usr/local/lib/arm-linux-gnueabihf: No such file or directory
ldconfig: Path `/lib/arm-linux-gnueabihf' given more than once
ldconfig: Path `/usr/lib/arm-linux-gnueabihf' given more than once
ldconfig: /lib/arm-linux-gnueabihf/ld-2.28.so is the dynamic linker, ignoring

ldconfig: Cannot stat /usr/lib/arm-linux-gnueabihf/libmca_common_ofi.so: No such file or directory
        libboost_fiber.so.1.67.0 -> libboost_fiber.so.1.67.0
        libboost_program_options.so.1.67.0 -> libboost_program_options.so.1.67.0
        libboost_system.so.1.67.0 -> libboost_system.so.1.67.0
        libboost_thread.so.1.67.0 -> libboost_thread.so.1.67.0
        libboost_unit_test_framework.so.1.67.0 -> libboost_unit_test_framework.so.1.67.0
        libboost_wave.so.1.67.0 -> libboost_wave.so.1.67.0
        libboost_iostreams.so.1.58.0 -> libboost_iostreams.so.1.58.0
        libboost_stacktrace_noop.so.1.67.0 -> libboost_stacktrace_noop.so.1.67.0
        libboost_stacktrace_backtrace.so.1.67.0 -> libboost_stacktrace_backtrace.so.1.67.0
        libboost_mpi_python27.so.1.67.0 -> libboost_mpi_python27.so.1.67.0
        libboost_math_tr1.so.1.67.0 -> libboost_math_tr1.so.1.67.0
        libboost_serialization.so.1.67.0 -> libboost_serialization.so.1.67.0
        libboost_graph_parallel.so.1.67.0 -> libboost_graph_parallel.so.1.67.0
        libboost_regex.so.1.67.0 -> libboost_regex.so.1.67.0
        libboost_locale.so.1.67.0 -> libboost_locale.so.1.67.0
        libboost_python27.so.1.67.0 -> libboost_python27.so.1.67.0
        libboost_timer.so.1.67.0 -> libboost_timer.so.1.67.0
        libboost_math_c99.so.1.67.0 -> libboost_math_c99.so.1.67.0
        libboost_log_setup.so.1.67.0 -> libboost_log_setup.so.1.67.0
        libboost_prg_exec_monitor.so.1.67.0 -> libboost_prg_exec_monitor.so.1.67.0
        libboost_chrono.so.1.67.0 -> libboost_chrono.so.1.67.0
        libboost_filesystem.so.1.67.0 -> libboost_filesystem.so.1.67.0
        libboost_math_c99f.so.1.67.0 -> libboost_math_c99f.so.1.67.0
        libboost_numpy27.so.1.67.0 -> libboost_numpy27.so.1.67.0
        libboost_random.so.1.67.0 -> libboost_random.so.1.67.0
        libboost_log.so.1.67.0 -> libboost_log.so.1.67.0
        libboost_math_tr1f.so.1.67.0 -> libboost_math_tr1f.so.1.67.0
        libboost_stacktrace_basic.so.1.67.0 -> libboost_stacktrace_basic.so.1.67.0
        libboost_signals.so.1.67.0 -> libboost_signals.so.1.67.0
        libboost_iostreams.so.1.67.0 -> libboost_iostreams.so.1.67.0
        libboost_context.so.1.67.0 -> libboost_context.so.1.67.0
        libboost_mpi.so.1.67.0 -> libboost_mpi.so.1.67.0
        libboost_stacktrace_addr2line.so.1.67.0 -> libboost_stacktrace_addr2line.so.1.67.0
        libboost_date_time.so.1.67.0 -> libboost_date_time.so.1.67.0
        libboost_wserialization.so.1.67.0 -> libboost_wserialization.so.1.67.0
        libboost_container.so.1.67.0 -> libboost_container.so.1.67.0
        libboost_type_erasure.so.1.67.0 -> libboost_type_erasure.so.1.67.0
        libboost_atomic.so.1.67.0 -> libboost_atomic.so.1.67.0
        libboost_coroutine.so.1.67.0 -> libboost_coroutine.so.1.67.0
        libboost_mpi_python37.so.1.67.0 -> libboost_mpi_python37.so.1.67.0
        libboost_graph.so.1.67.0 -> libboost_graph.so.1.67.0
        libboost_python37.so.1.67.0 -> libboost_python37.so.1.67.0
        libboost_numpy37.so.1.67.0 -> libboost_numpy37.so.1.67.0
ldconfig: /lib/ld-linux.so.3 is the dynamic linker, ignoring

mav@maverick-raspberry:/usr/local/lib$ 
```

SamuelDudley, 2020-08-17:
okay, new problem
```
mav@maverick-raspberry:/usr/local/lib$ ~/software/maverick/bin/rosmaster.sh apdev
... logging to /srv/maverick/var/log/ros/apdev/d5d17920-e06a-11ea-8826-b827eb2b499a/roslaunch-maverick-raspberry-10609.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://maverick-raspberry:36015/
ros_comm version 1.15.8

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES

auto-starting new master
process[master]: started with pid [10618]
ROS_MASTER_URI=http://maverick-raspberry:6114/

setting /run_id to d5d17920-e06a-11ea-8826-b827eb2b499a
process[rosout-1]: started with pid [10629]
started core service [/rosout]
/srv/maverick/software/ros/current/lib/rosout/rosout: error while loading shared libraries: liblog4cxx.so.10: cannot open shared object file: No such file or directory
[rosout-1] process has died [pid 10629, exit code 127, cmd /srv/maverick/software/ros/current/lib/rosout/rosout __name:=rosout __log:=/srv/maverick/var/log/ros/apdev/d5d17920-e06a-11ea-8826-b827eb2b499a/rosout-1.log].
log file: /srv/maverick/var/log/ros/apdev/d5d17920-e06a-11ea-8826-b827eb2b499a/rosout-1*.log
[rosout-1] restarting process
```

`liblog4cxx.so.10`

SamuelDudley, 2020-08-17:
`sudo apt-get install liblog4cxx-dev -y`

```
mav@maverick-raspberry:/usr/local/lib$ sudo ldconfig -v | grep cxx
ldconfig: Can't stat /usr/local/lib/arm-linux-gnueabihf: No such file or directory
ldconfig: Path `/lib/arm-linux-gnueabihf' given more than once
ldconfig: Path `/usr/lib/arm-linux-gnueabihf' given more than once
ldconfig: /lib/arm-linux-gnueabihf/ld-2.28.so is the dynamic linker, ignoring

ldconfig: Cannot stat /usr/lib/arm-linux-gnueabihf/libmca_common_ofi.so: No such file or directory
        liblog4cxx.so.10 -> liblog4cxx.so.10.0.0
        libmpi_cxx.so.40 -> libmpi_cxx.so.40.10.0
ldconfig: /lib/ld-linux.so.3 is the dynamic linker, ignoring
```
winning yet?
```
mav@maverick-raspberry:/usr/local/lib$ ~/software/maverick/bin/rosmaster.sh apdev
... logging to /srv/maverick/var/log/ros/apdev/6b3cd540-e06b-11ea-a5e4-b827eb2b499a/roslaunch-maverick-raspberry-11977.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://maverick-raspberry:33717/
ros_comm version 1.15.8

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES

auto-starting new master
process[master]: started with pid [12009]
ROS_MASTER_URI=http://maverick-raspberry:6114/

setting /run_id to 6b3cd540-e06b-11ea-a5e4-b827eb2b499a
process[rosout-1]: started with pid [12024]
started core service [/rosout]
/srv/maverick/software/ros/current/lib/rosout/rosout: error while loading shared libraries: libconsole_bridge.so.0.4: cannot open shared object file: No such file or directory
[rosout-1] process has died [pid 12024, exit code 127, cmd /srv/maverick/software/ros/current/lib/rosout/rosout __name:=rosout __log:=/srv/maverick/var/log/ros/apdev/6b3cd540-e06b-11ea-a5e4-b827eb2b499a/rosout-1.log].
log file: /srv/maverick/var/log/ros/apdev/6b3cd540-e06b-11ea-a5e4-b827eb2b499a/rosout-1*.log
```
no...

SamuelDudley, 2020-08-17:
```
mav@maverick-raspberry:/usr/local/lib$ sudo apt-get install libconsole-bridge-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libconsole-bridge0.4
The following NEW packages will be installed:
  libconsole-bridge-dev libconsole-bridge0.4
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 14.8 kB of archives.
After this operation, 64.5 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://raspbian.mirror.digitalpacific.com.au/raspbian buster/main armhf libconsole-bridge0.4 armhf 0.4.3+dfsg-1 [7,388 B]
Get:2 http://raspbian.melbourneitmirror.net/raspbian buster/main armhf libconsole-bridge-dev armhf 0.4.3+dfsg-1 [7,420 B]
Fetched 14.8 kB in 5s (3,097 B/s)                                   
Selecting previously unselected package libconsole-bridge0.4:armhf.
(Reading database ... 180089 files and directories currently installed.)
Preparing to unpack .../libconsole-bridge0.4_0.4.3+dfsg-1_armhf.deb ...
Unpacking libconsole-bridge0.4:armhf (0.4.3+dfsg-1) ...
Selecting previously unselected package libconsole-bridge-dev:armhf.
Preparing to unpack .../libconsole-bridge-dev_0.4.3+dfsg-1_armhf.deb ...
Unpacking libconsole-bridge-dev:armhf (0.4.3+dfsg-1) ...
Setting up libconsole-bridge0.4:armhf (0.4.3+dfsg-1) ...
Setting up libconsole-bridge-dev:armhf (0.4.3+dfsg-1) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
mav@maverick-raspberry:/usr/local/lib$ sudo ldconfig -v | grep _bridge
ldconfig: Can't stat /usr/local/lib/arm-linux-gnueabihf: No such file or directory
ldconfig: Path `/lib/arm-linux-gnueabihf' given more than once
ldconfig: Path `/usr/lib/arm-linux-gnueabihf' given more than once
ldconfig: /lib/arm-linux-gnueabihf/ld-2.28.so is the dynamic linker, ignoring

ldconfig: Cannot stat /usr/lib/arm-linux-gnueabihf/libmca_common_ofi.so: No such file or directory
        libconsole_bridge.so.0.4 -> libconsole_bridge.so.0.4
ldconfig: /lib/ld-linux.so.3 is the dynamic linker, ignoring

mav@maverick-raspberry:/usr/local/lib$ 
```

```
mav@maverick-raspberry:/usr/local/lib$ ~/software/maverick/bin/rosmaster.sh apdev
... logging to /srv/maverick/var/log/ros/apdev/bcb166a2-e06b-11ea-95f7-b827eb2b499a/roslaunch-maverick-raspberry-12245.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://maverick-raspberry:45553/
ros_comm version 1.15.8

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.8

NODES

auto-starting new master
process[master]: started with pid [12254]
ROS_MASTER_URI=http://maverick-raspberry:6114/

setting /run_id to bcb166a2-e06b-11ea-95f7-b827eb2b499a
process[rosout-1]: started with pid [12265]
started core service [/rosout]
```
🥳

SamuelDudley, 2020-08-17:
summary:
`sudo apt-get install libconsole-bridge-dev liblog4cxx-dev ibboost1.67-all -y`

fnoop, 2020-08-17:
Very odd, shoudln't ever have to install these by hand.  Must be missing a rosdep somewhere, not seen this before.

fnoop, 2020-08-17:
Ah balls I know what the problem is - I took a shortcut creating the maverick-raspberry image by copying maverick-raspberrylite ~/software to cut down on the build time.  This bypasses the various system installs that are done inside the various protected build flag compiles which aren't then done after the build flags are installed.

I'll fix and produce a new image.  The raspberrylite image should be fine :)

SamuelDudley, 2020-08-18:
I flashed the lite image and that appears to be working.
I will get around to looking at this api issue :)

fnoop, 2020-08-18:
I've fixed/updated the full raspberry image!

ds-master, 2020-09-11:
Running `maverick start api@fc` produces the following error:

```
Traceback (most recent call last):
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick-api.py", line 40, in <module>
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     api.initialize()
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 47, in initialize
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     generate_schema()
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/__init__.py", line 129, in generate_schema
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     f"{__name__}.{module_folder_name}.{module_name}"
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/python/lib/python3.7/importlib/__init__.py", line 127, in import_module
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     return _bootstrap._gcd_import(name[level:], package, level)
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/__init__.py", line 2, in <module>
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     from maverick_api.modules.api.mavros.mavros_connection import MAVROSConnection
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/mavros_connection.py", line 21, in <module>
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     from maverick_api.modules.api.mavros.mavros_param import ParamInterface
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/api/mavros/mavros_param.py", line 42, in <module>
Sep 11 06:51:45 maverick-raspberry api.sh[11685]:     from graphql.error import INVALID
Sep 11 06:51:45 maverick-raspberry api.sh[11685]: ImportError: cannot import name 'INVALID' from 'graphql.error' (/srv/maverick/software/python/lib/python3.7/site-packages/graphql/error/__init__.py)
```
How can this be fixed? Followed this thread but could not find a viable solution. Ran `maverick self-update` but same issue persists.

SamuelDudley, 2020-09-11:
Hi, yep this is still an issue that I have not yet worked on. You can modify the configuration file for maverick-api to avoid loading the mavros plugin for the time being.

SamuelDudley, 2020-09-11:
I'll make this change in maverick tonight:
https://github.com/goodrobots/maverick/blob/stable/manifests/maverick-modules/maverick_web/templates/maverick-api.conf.erb#L44

That line needs `mavros` in the currently empty list.

ds-master, 2020-09-14:
Hi, thanks that solves this problem. However, I'm getting another issue:
```
Traceback (most recent call last):
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:   File "/srv/maverick/software/maverick-api/maverick-api.py", line 40, in <module>
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:     api.initialize()
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:   File "/srv/maverick/software/maverick-api/maverick_api/modules/base/apiserver/__init__.py", line 70, in initialize
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:     server.listen(port=options.server_port_nonssl, address=options.server_interface)
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:   File "/srv/maverick/software/python/lib/python3.7/site-packages/tornado/tcpserver.py", line 151, in listen
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:     sockets = bind_sockets(port, address=address)
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:   File "/srv/maverick/software/python/lib/python3.7/site-packages/tornado/netutil.py", line 174, in bind_sockets
Sep 14 06:02:34 maverick-raspberry api.sh[2343]:     sock.bind(sockaddr)
Sep 14 06:02:34 maverick-raspberry api.sh[2343]: OSError: [Errno 98] Address already in use
```
I think my configuration is wrong somewhere. Could you please help me with this?

SamuelDudley, 2020-09-14:
They modified the graphql core library, specifically around the custom type handling. The original root error you ran into is some code designed to handle the parameter value type which can be either a float or an int.

Anyway, it looks like there is another issue possibly with the port default port selection. I'll take a look after work today.